fr3ts0n/AndrOBD #220: Data readout priority

gnu-xv, 2022-11-08:
The data readout period depends on the the number of items selected.
The more items are selected the longer is the readout, and the lower is the update frequency.
This behavior is not ideal for accurate engine monitoring .
In fact, some obd data vary slowly and thus don't need to be read frequently, e.g. oil or coolant temperatures.
In contrast, others vary  rapidly and thus should be read as often as possible, eg. rpm, speed or engine load.

Would it be possible to implement some sort of readout priority?
For example, oil temperature could be read every 10th loop iteration, whereas rpm and speed are read at every loop iteration.
The priority could simply be a number (x), and a data is only read when the iteration number modulo the priority is zero (or any fixed number, between 0 and x-1).
In the csv export, you can either output the last known value or (better) an empty value (-), when a data hasn't been read.

This readout with priority could be enabled or disabled through configuration menu.
The individually priorities could be defined in the file pids.csv and, ideally, editable.

fr3ts0n, 2022-11-12:
I agree, but I would rather use a physical update timing value than any iteration counter / priority. 
- It needs to be considered, that 1 OBD PID may contain multiple data items. 
  - > Highest frequency item should re-trigger the PID request.
- CSV seems be a good place for config.
- Can be part of data costomisation

gnu-xv, 2022-11-12:
For sure, a physical update timing value (like a lower limit of update period) would be nice.
The idea with the counter/priority was to keep it very simple and see if it improves timing.

fr3ts0n, 2022-11-26:
I described the feature / changes here:
https://github.com/fr3ts0n/AndrOBD/wiki/Data-item-update-time#pid-prioritization

Test version will follow soon ...

fr3ts0n, 2022-11-26:
Test version is available here:
- https://t.me/AndrOBD_dev/247

closed 2023-12-20 by fr3ts0n