fr3ts0n/AndrOBD #221: "Data to visualize" overview/possible overhaul

gianniqq, 2022-11-30:
**Is your feature request related to a problem? Please describe.**
Currently the application allows the user to show the possible pieces of information (are they called PIDs?) by going to Settings->OBD Settings -> Data to visualize; here a scrollable popup window is shown with a long list of checkable items.
At least in my experience the list is quite long and hard to navigate, further more it is not clear to me how this list is generated.
By that i mean: does the shown item actually represent something that the ECU(s) can provide, or is the list a list of all the possible data that the ECUs may or may not provide. I checked a lot of items but only some of them actually show on the "OBD Data" section.
Furthermore, it looks like there are possibly duplicate items? Any information regarding this topic would be kindly apreciated.

**Describe the solution you'd like**
I think it would be nice to have a dedicated Activity where one can filter the available items to quickly show relevant matches (eg. I want data related to "fuel", I type in "fuel" and all and only the items with "fuel" in their name show).

I am also very interested in some answers to my questions regarding the provided list of PIDs (?).

Here is a screenshot of the current way the list is shown:
[image]

Thank you for your dedication, this is a cool project.

fr3ts0n, 2022-11-30:
Hi @gianniqq ,
There is already a similar request concerning this data filtering setting. (Issue #205)
I have been checking into this earlier and also came to the conclusion that this will have to be changed to a separate activity rather than just a preference setting. Unfortunately preference settings do not offer any list filtering / search / select all mechanism.

To answer your questions on the list content:
- The list is generated of all possible OBD data items (PID's) a vehicle might support- or not.
- There are several PID's defined in the OBD spec which are of the same meaning, but for example have a different data range/resolution. So the list of textual descriptions may have double entries, but these entries represent different OBD PID's.

gianniqq, 2022-11-30:
Hi,
I completely missed the existing issue, sorry about that.

Thank you for answering my questions!
So it's a list of all possible available (or not) PIDs.
I suppose it would be possible to detect what PIDs are actually available on a vehicle (maybe) since when I select one of them sometimes it isn't shown (-> meaning it's most likely not supported). That would be extremely useful during the filtering process too!
It must depend on the way the protocol responds when it's asked a non-available PID.

fr3ts0n, 2022-12-01:
The settings dialog is based on the list of ALL known PIDs of OBD.
The filtering there is global, and NOT dependent on any vehicle.
-Any PID which is disabled in settings will NOT be shown on any vehicle

On connection to vehicle the app asks the vehicle which PID's it supports.
- All PIDs which are supoorted by vehicle AND enabled in filter setting will be shown in OBD data list

Now you can select your preferred items for this vehicle / session (by long pressing) and apply dynamic filter for
  - OBD data list
  - Gauges list
  - Graph charting

This dynamic filter is intended to limit the data updates to the items of your interest only, which results in faster data updates of selected items.

closed 2023-12-20 by fr3ts0n