fr3ts0n/AndrOBD #224: Bug report

monikamishra-54, 2022-12-28:
# Describe the bug

Shows OBD protocol error when clicked on clear fault code.

## AndrOBD Debug log files

E/com.fr3ts0n.prot: (NRC:0x22) Conditions not correct (SVC:0x04)

## Additional context

Screenshot is attached for more reference.
[
[image]
](url)

fr3ts0n, 2022-12-28:
This is NOT a bug in AndrOBD.
This is your car reporting that clearing fault codes is NOT allowed in current situation.
"Conditions not correct" means that not all conditions are met to perform the requested action.

Did you have your engine running while trying to clear codes?
- I know of several vehicles which don't allow this while the engine is running.

Please try to clear codes with engine OFF.

monikamishra-54, 2022-12-29:
Thank you, It worked when engine is OFF but now it shows response is pending when I clear fault code
E/com.fr3ts0n.prot: (NRC:0x78) Request correctly received  but response is pending
[image]

fr3ts0n, 2022-12-29:
Could you please provide a logfile (FINE) for that?
See: https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues#debug-log-files

fr3ts0n, 2022-12-29:
Sorry, didn't mean to close it yet ...

monikamishra-54, 2022-12-29:
Didn't have FINE log level in my android device only debug and verbose is there
[image]

fr3ts0n, 2022-12-29:
Your screenshot above confirms the symptom, but does not provide any details what did lead to the symptom.
Sorry, but any further analysis of this does require a log output with higher granularity.

closed 2023-12-20 by fr3ts0n