fr3ts0n/AndrOBD #226: difference between OBD fault code, OBD fault codes(Pending) and OBD fault code(permanent)

monikamishra-54, 2022-12-29:
**Is your feature request related to a problem? Please describe.**
Please explain difference between OBD fault code, OBD fault codes(Pending) and OBD fault code(permanent) and what fault code should we consider while testing.

**Describe alternatives you've considered**
can we add all three type of fault code in one screen

fr3ts0n, 2022-12-29:
DTC's are specified by OBD in 3 different services:
- 0x03: DTCs
- 0x07: Pending DTC's 
- 0x0A: Permanent DTC's
Each type of DTC's is to be requested separately
(See also: https://en.wikipedia.org/wiki/OBD-II_PIDs)

The latest test version of AndrOBD already combines all 3 different service requests into 1 list

monikamishra-54, 2022-12-29:
Thanks for response, but I'm not able to find 3 different service requests into 1 list in code.

fr3ts0n, 2022-12-29:
You can find test versions here:
- https://t.me/AndrOBD_dev
(See also:  https://github.com/fr3ts0n/AndrOBD#readme)

closed 2023-01-14 by fr3ts0n