fr3ts0n/AndrOBD #228: Confirmed fault code

monikamishra-54, 2022-12-29:
**Is your feature request related to a problem? Please describe.**
Is app providing confirmed fault code

**Describe the solution you'd like**
confirmed and permanent fault code is same.

**Describe alternatives you've considered**
https://support.bluedriver.com/support/solutions/articles/43000551802-confirmed-pending-permanent-and-enhanced-codes#:~:text=Once%20the%20sensor%20starts%20to,a%20pending%20code%20will%20appear.&text=A%20confirmed%20code%20indicates%20that,an%20issue%20requiring%20your%20attention

This article shows three types of fault code

fr3ts0n, 2022-12-29:
NO, permanent != confirmed
- Pending codes did occur at least once within the drive cycle, but not long-/often enough to be confirmed yet
- Confirmed codes did occur multiple times within the last drive cycles, but may be cleared by "Clear DFC" service.
  - If problem still occurs, DFC will be activated again
- Permanent codes may not be cleared by "Clear DFC" service. 
  - They may be resolved in a garage by parts change only.

monikamishra-54, 2022-12-29:
Thank you for quick response,
Sorry for silly questions but I'm new in this field my question is app is providing all types of fault code i.e Powertrain, Chassis, Body
[image]

fr3ts0n, 2022-12-29:
See current supported code list: https://github.com/fr3ts0n/AndrOBD/blob/master/library/src/main/java/com/fr3ts0n/ecu/prot/obd/res/codes.properties

However: Above list is the superset of known codes. You will only see the codes which the vehicle supports on OBD protocol.

monikamishra-54, 2022-12-30:
Thank You, but this means app doesn't support Chassis and body code?

fr3ts0n, 2022-12-30:
Yes, that is what OBD protocol is designed for: Emissions related diagnostics!

monikamishra-54, 2022-12-30:
Thanks for response, Can you help me which library or any github open source code or any idea how we can achieve this for getting  Chassis and body code from OBD device. Is there way we can extent this code to get other DTC services.
"I tried everywhere but didn't find anything related this. Please Help "

fr3ts0n, 2022-12-31:
For implememting NON OBD codes / data you need a lot of:
- OEM specific documentation
- Time / equipment for reverse engineering
See also: https://github.com/fr3ts0n/AndrOBD/wiki/Frequently-asked-questions#will-androbd-support-additional-non-obd-sensor-data
Good luck

closed 2023-01-14 by fr3ts0n