fr3ts0n/AndrOBD #229: Service 0x0A not supported

monikamishra-54, 2022-12-30:
# Describe the bug

OBD protocol error

## Expected behavior

Torque app showing P0532 error but in AndrOBD app shows Service 0x0A not supported

## AndrOBD Debug log files
[AndrOBD.log.3.txt](https://github.com/fr3ts0n/AndrOBD/files/10323366/AndrOBD.log.3.txt)

## Additional context

Screenshot of Torque app
[image]

fr3ts0n, 2022-12-30:
Pls attach Log Files with protokol (FINE) information. 
The only thing I can see in the file is that you switched between modes and Mode 10 ist rejected...

monikamishra-54, 2022-12-30:
[AndrOBD.log.0 (1).txt](https://github.com/fr3ts0n/AndrOBD/files/10323560/AndrOBD.log.0.1.txt)

monikamishra-54, 2022-12-30:
Hello sir ,
we did not clear DFC in whole process.
also is your code does not support DTC services ?

fr3ts0n, 2022-12-30:
In above log file you can see that the ECU rejects a request to service 0x0A with NFC 0x11 (Not supported)
The log does NOT contain any request to other DTC services

AndrOBD does support all three services specified by OBD concerning DTC reading (0x3, 0x7, 0xA) and one for clearing DTC's (0x4)

As mentioned above, the torque app may also read other, non OBD services, which AndrOBD does NOT.

monikamishra-54, 2022-12-30:
AndrOBD does support all three services specified by OBD concerning DTC reading (0x3, 0x7, 0xA) and one for clearing DTC's (0x4)
Said above it should support 0xA service but after clearing fault code its still showing obd protocol error
[image]

fr3ts0n, 2022-12-31:
This is not the app lacking support.
This is YOUR VEHICLE reporting the lack of support!

closed 2023-01-14 by fr3ts0n