fr3ts0n/AndrOBD #230: Impetial Units seem to be measured correctly but show up as metric

tubegeek, 2023-01-14:
When Imperial units are selected, gauges and charts retain the metric labels for units. It appears that the measurements are correct though - examples - coolant temp 180 is shown as C, but 180 seems right for F. I drove 20 Mph per my dash speedometer, it was displayed as 20 Kph.

fr3ts0n, 2023-01-14:
Hi @tubegeek ,
Thanks for your report. 

Problem confirmed. 
It's amazing how things like that stay undetected for quite some time ...

Cause is identified. 
Fixed test version will come soon.

tubegeek, 2023-01-14:
My pleasure. AndrOBD is working quite well for me, using the cheapest available bluetooth ELM327 adapter. Less than 6 $USD on eBay.

Thanks for the effort!

fr3ts0n, 2023-01-14:
Test version (V2.6.3) is now available here:
https://t.me/AndrOBD_dev/250

closed 2023-01-14 by fr3ts0n