fr3ts0n/AndrOBD #231: Add support for electric cars/protocol

Hausmeister0815, 2023-03-02:
Hi,
first thanks for your work. :)

It would be great if the app could be used with electric cars.
I havnt any expirience with OBD, so sorry if i am wrong.

On first test with an opel corsa-e i could connect to the elm327 but no connection to car.

I think that they use another protocol not supported. Or maybe only a PSA related protocol addition?
Because of this i dont added logs now. If you think it is helpfull anyway, is there a prefered log level for you?

Testing with another app it is connecting but shows only "Opel OBD II / EOBD Electric / Hybrids (PSA based)".

Searching for this keywords is not helpfull.

It looks like the base is the iso-15765-4  (11bit 500kb), but up to now dont found a list with the relevant pids.
Most detailed information i found is this.
https://www.goingelectric.de/forum/viewtopic.php?p=1514148#p1514148

What do you think, is it possible?
For testing and log creation i will do my best to support.

Best Regards 
Tom

fr3ts0n, 2023-03-15:
Hi Tom,
Technically it definitely is possible. 
The ELM Adapters and communication library of AndrOBD would even support the required UDS based services.
However, the thing is that as soon as you leave the standardized path of OBD protocol, you drastically increase the effort of analysis of documentation (best case) or reverse engineering (worst case).
This effort would definitely exceed the limits of a spare time, free of charge, OSS project.

I have also explained this here in the [FAQ](https://github.com/fr3ts0n/AndrOBD/wiki/Frequently-asked-questions#will-androbd-support-additional-non-obd-sensor-data)

Hausmeister0815, 2023-03-15:
Hi,
i understand your point and agree. Of course, the part of reverse engineeing will be on my side if there are non standard pids.

But anyway, i think the basic OBD protocoll should be able to connect to the car, even if there is only e.g. the 12V battery voltage available. Or read out error codes. Isn't it defined by the OBD standard?

I made a log and it seems that there are some commands like ATPC are answered with ok but later at 0900 "unable to connect". 
Havent checked the source code (same problem with spare time) and not sure what 0900 mean. In settings it is not listed at the disable ELM command page.

Test case.
Car ready -> Start App.
Connect -> Vehicle Info -> Wait a few seconds -> disconnect.
Connect -> OBD Data -> wait -> disconnect 
Connect -> OBD fault codes -> wait -> disconnect -> close app.
[AndrOBD.log.0.txt](https://github.com/fr3ts0n/AndrOBD/files/10984405/AndrOBD.log.0.txt)

fr3ts0n, 2023-03-17:
Hi,
It really looks  like there is no answer of your car from any ECU on the network to any requested OBD service.
Communication with the ELM adapter looks fine, but the car just plainly doesn't answer OBD protocol.

The "0900" does request OBD service 0x09, PID 0x00 (Get List of supported PIDs (0-31)  of OBD vehicle information)
(see https://en.wikipedia.org/wiki/OBD-II_PIDs)

Based on your link above it looks like the vehicle does UDS protocol and reports data on service 22 (Read data by identifier)
(see: https://en.wikipedia.org/wiki/Unified_Diagnostic_Services)

Tuoris, 2025-02-11:
@Hausmeister0815 , I had the same idea, but seems like EVs does not always conform to the OBD2 standard simply because it (standard) was designed to comply with emissions standards [1].

I've did some digging, data logging and testing with my Huyndai Kona using AndrOBD and my own app [2] and found out that 
it does not support most OBD2 pids (with few exceptions). Most of the data is available with UDS Service ID 22, as @fr3ts0n stated.
The data itself is in completely different format from the PSA based cars.

FYI: the ok response from `ATPC`command (AT Protocol close) simply mean the OBD2 adapter is working ok - no car data queries are performed with such commands. (again, this confirms @fr3ts0n words). I have a few OBD2 adapters and most of them properly respond to `AT Z`, `AT WS` and other AT commands, but can not get the data from the car itself (only Vgate iCar2 works [3]).

[1] https://en.wikipedia.org/wiki/On-board_diagnostics
[2] https://tuoris.github.io/elm327_debugger/en/ , source code https://github.com/Tuoris/elm327_debugger
[3] https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/issues/41#issuecomment-2651687593