fr3ts0n/AndrOBD #232: KW1281 support in sources

mvglasow, 2023-03-14:
According to the FAQ, non-OBD protocols are not supported for valid reasons.

However, there is this file in the source code: https://github.com/fr3ts0n/AndrOBD/blob/master/library/src/main/java/com/fr3ts0n/ecu/prot/vag/Kw1281Prot.java

This looks like an implementation of the Volkswagen Group’s KW1281 protocol, although after a quick glimpse it does not appear to be used anywhere in the app.

How mature is this code? Does it work with an ELM327 adapter (if not, what type of adapter was it written for)?

(Not a feature request, but if this code works, I would like to build on it for my own project.)

fr3ts0n, 2023-03-15:
Yes, this file is a implementation of VAG KW1281. It is not part of the app, but is contained in the generic java library
There is a JRE GUI application in the repo (com/fr3ts0n/ecu/gui/application/VagTestFrame.java) which actually uses it.
It can be used with a USB KKL adapter. I used it to diagnose my old Skoda Fabia.

mvglasow, 2023-03-16:
Such a gem hidden deep down in the code :-) There didn’t seem to be a build recipe, so I pieced together a POM, based on https://github.com/fr3ts0n/AndrOBD/blob/master/library/build.gradle. A few fixes were needed:

* Add JFreeChart as a dependency
* Install RXTX, add it as a dependency and re-add missing imports in `SerialConfigPanel` (which got removed in 59795d6d).
* Also in `SerialConfigPanel`, import `java.util.logging.Logger`, fix the calls to `getLogger` (the JDK version takes a class name as a string, but not a class) and change `warn` to `warning`. (This code was written for Log4J, 59795d6d removed the import but left the method calls in place).

This eliminated most of the build errors, with only one remaining: In `PvTreeNode#findChild(Object)`, a local variable of type `Iterator<PvTreeNode>` is assigned `children.iterator()`, which is of class `Iterator<TreeNode>`. This seems to have been there since the initial commit. What went wrong here?

mvglasow, 2023-03-16:
...however, I am getting an `UnsatisfiedLinkError` because `SerialExt.java` tries to load a native library named `SerialExt`. Is that part of your code? Is it anywhere to be found in this repository?

fr3ts0n, 2023-03-17:
Oh yes, there was a lib to it to allow communication on odd baud rates like 10400 ...
I'll have a look to find it again ...

VasilKotsov, 2025-01-04:
Hello I'm looking of app support this kw1281 protocol. Is there any way to add a support of it? I have a Bluetooth elm327 adapter that had a support of it. It will be nice if you add it. Thank you.

VasilKotsov, 2025-09-28:
Hello please add support for KW1281, as we can see there is [code](https://github.com/fr3ts0n/AndrOBD/blob/master/library/src/main/java/com/fr3ts0n/ecu/prot/vag/Kw1281Prot.java) inside your repo, so please add it. I will be really grateful and appreciated if you do it. Thanks for your hard work.