fr3ts0n/AndrOBD #233: Some new cheap ELM327 adapters work with other apps like Torque but not AndrOBD (BLE/BT 4.x related?)

DevelLevel, 2023-03-20:
# Describe the bug

After connecting to the adapter in Android Bluetooth settings successfully and pressing on connect in the top right corner of AndrOBD a toast comes up saying "Connected to OBD II", instantly followed by another toast saying "Device disconnected".
This is the same behavior as in [issue 64](https://github.com/fr3ts0n/AndrOBD/issues/64), though it can't be resolved with disabling the ATZ command...

## To Reproduce

Steps to reproduce the behavior:
1. Plug the adapter into the OBD 2 port and turn the ignition on
2. Connect to the Bluetooth device in Android settings (In my case with the password: "1234")
3. Open AndrOBD, press connect and select the ELM327 Bluetooth device
4. Observe the error

## Expected behavior

1. Plug the adapter into the OBD 2 port and turn the ignition on
2. Connect to the Bluetooth device in Android settings (In my case with the password: "1234")
3. Open AndrOBD, press connect and select the ELM327 Bluetooth device
**4. Get a established connection for a prolonged time and the ability to read data.**

## AndrOBD Debug log files
<details>
<summary>Log without the ATZ command disabled</summary>
<pre><code>2023-03-19	16:20:36.590	INFO	AndrOBD	AndrOBD V2.5.5 starting
2023-03-19	16:20:36.642	INFO	AndrOBD	Conversion: metric
2023-03-19	16:20:42.479	INFO	CommService	BT socket - UUIDs:00001101-0000-1000-8000-00805f9b34fb,
2023-03-19	16:20:42.480	INFO	CommService	BEGIN mBtConnectThread SocketType:Insecure
2023-03-19	16:20:43.261	INFO	CommService	BEGIN mBtWorkerThread
2023-03-19	16:20:43.262	INFO	stream	RX Thread started
2023-03-19	16:20:43.268	WARNING	stream	com.fr3ts0n.prot.StreamHandler@9ac83e3 RX: End of stream!
2023-03-19	16:20:43.269	INFO	stream	RX Thread stopped
2023-03-19	16:20:43.269	INFO	CommService	Closing BT socket</code></pre></details>

<details>
<summary>Log when disabling various commands such as ATZ</summary>
<pre><code>2023-03-19	16:46:33.416	INFO	AndrOBD	AndrOBD V2.5.5 starting
2023-03-19	16:46:33.417	FINER	com.fr3ts0n.pvs	+PvListener:com.fr3ts0n.pvs.PvList[null]->com.fr3ts0n.ecu.gui.androbd.MainActivity@bf0ec3f
2023-03-19	16:46:33.417	FINER	com.fr3ts0n.pvs	+PvListener:com.fr3ts0n.pvs.PvList[null]->com.fr3ts0n.ecu.gui.androbd.MainActivity@bf0ec3f
2023-03-19	16:46:33.417	FINER	com.fr3ts0n.pvs	+PvListener:com.fr3ts0n.pvs.PvList[null]->com.fr3ts0n.ecu.gui.androbd.MainActivity@bf0ec3f
2023-03-19	16:46:33.418	FINER	com.fr3ts0n.pvs	+PvListener:com.fr3ts0n.pvs.PvList[null]->com.fr3ts0n.ecu.gui.androbd.MainActivity@bf0ec3f
2023-03-19	16:46:33.423	FINE	AndrOBD	Adapter: android.bluetooth.BluetoothAdapter@8bec2b5
2023-03-19	16:46:33.504	INFO	AndrOBD	Conversion: metric
2023-03-19	16:46:40.773	FINE	BtDeviceListActivity	Sending Result...
2023-03-19	16:46:40.811	FINE	CommService	connect to: 00:10:CC:4F:36:03
2023-03-19	16:46:40.812	FINE	CommService	setState() NONE -> CONNECTING
2023-03-19	16:46:40.817	INFO	CommService	BT socket - UUIDs:00001101-0000-1000-8000-00805f9b34fb,
2023-03-19	16:46:40.819	INFO	CommService	BEGIN mBtConnectThread SocketType:Insecure
2023-03-19	16:46:40.820	FINE	CommService	Connect BT socket
2023-03-19	16:46:40.942	FINE	CommService	connected, Socket Type:Insecure
2023-03-19	16:46:40.942	FINE	CommService	create BtWorkerThread: Insecure
2023-03-19	16:46:40.943	INFO	CommService	BEGIN mBtWorkerThread
2023-03-19	16:46:40.943	INFO	stream	RX Thread started
2023-03-19	16:46:40.944	FINE	CommService	setState() CONNECTING -> CONNECTED
2023-03-19	16:46:40.953	WARNING	stream	com.fr3ts0n.prot.StreamHandler@d59e04f RX: End of stream!
2023-03-19	16:46:40.953	INFO	stream	RX Thread stopped
2023-03-19	16:46:40.953	FINE	CommService	stop
2023-03-19	16:46:40.954	INFO	CommService	Closing BT socket
2023-03-19	16:46:40.954	FINE	CommService	setState() CONNECTED -> OFFLINE
2023-03-19	16:46:40.955	FINE	CommService	setState() OFFLINE -> OFFLINE
2023-03-19	16:46:40.961	FINER	com.fr3ts0n.pvs	PvChange:32:null=null
2023-03-19	16:46:40.961	FINER	com.fr3ts0n.pvs	Notify:com.fr3ts0n.ecu.gui.androbd.MainActivity@bf0ec3f=33
2023-03-19	16:46:40.962	FINER	com.fr3ts0n.pvs	PvChange:32:null=null
2023-03-19	16:46:40.962	FINER	com.fr3ts0n.pvs	Notify:com.fr3ts0n.ecu.gui.androbd.MainActivity@bf0ec3f=33
2023-03-19	16:46:40.962	FINER	com.fr3ts0n.pvs	PvChange:32:null=null
2023-03-19	16:46:40.962	FINER	com.fr3ts0n.pvs	Notify:com.fr3ts0n.ecu.gui.androbd.MainActivity@bf0ec3f=33
2023-03-19	16:46:40.963	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@e9abc99 TX:'ATZ'
2023-03-19	16:46:40.963	FINER	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@e9abc99 TX:41 54 5A  : ATZ
2023-03-19	16:47:40.425	FINE	com.fr3ts0n.prot	ELM command 'ATZ' -> disabled
2023-03-19	16:47:40.426	FINE	com.fr3ts0n.prot	ELM command 'ATWS' -> enabled
2023-03-19	16:47:40.426	FINE	com.fr3ts0n.prot	ELM command 'ATPC' -> enabled
[... 141 lines cut ...]
2023-03-19	16:50:34.212	FINE	com.fr3ts0n.prot	ELM command 'ATAT' -> enabled
2023-03-19	16:50:34.212	FINE	com.fr3ts0n.prot	ELM command 'ATST' -> enabled
2023-03-19	16:50:34.212	FINE	com.fr3ts0n.prot	ELM command 'ATSH' -> enabled
2023-03-19	16:50:34.213	FINE	com.fr3ts0n.prot	ELM command 'ATCRA' -> enabled
2023-03-19	16:50:34.213	FINE	com.fr3ts0n.prot	ELM command 'ATCRA' -> enabled</code></pre></details>

### Log detail level

INFO/FINER

## Important details on your environment:

Tested on
- Android OS version: 13 (LineageOS) & 10 (Stock Samsung)
- AndrOBD version: V2.5.5 (and some older)
- OBD adapter type: Cheap ELM327 mini interface
- Connected vehicle make, type, year, engine size: [VW, Golf, 2006, 1.6] & [VW, Passat, 2016, 1.4]

## Additional context

The adapter seems to be a v2.1 clone that after my research seems to be capable of BLE, which could be a problem if it is not backward compatible?
It was bought on [CDON](https://cdon.se/fordon/bluetooth-felkodslasare-obd2-elm327-bildiagnostik-p36716638). But the seller does not provide a lot of information. The device is though very similar to [this one](https://www.amazon.com/ELM327-Bluetooth-Scanner-Diagnostic-Android/dp/B09B2Q7WJS) which specifies that "make sure to select in settings, "Connection type": Bluetooth 4.0 (LE)".
As mentioned by @fr3ts0n in the telegram group this might require a "BLE specific comms module" or a command to enable BT version 1.x, 2.x or 3.x mode of some sort...

Any thoughts?

fr3ts0n, 2023-03-25:
I do have a similar / equal adapter available (also reports ELM V1.5), but unfortunately I can not reproduce the problem with this.

The only obvious thing I could find in the HCI Snoop files is some difference in the signal timing:
- In the Carscanner log there seems to be a delay of approx. 500 ms between connection acknowledgement and 1st command
- AndrOBD does not wait and sends the 1st command after approx. 15 ms

There is a chance that the adapter needs some preparation time when it can not take any command too early after connection is established.
I will create a test version which includes a delay like this ...

fr3ts0n, 2023-03-25:
Test version V2.6.4 is available here:
https://t.me/AndrOBD_dev/251

DevelLevel, 2023-03-26:
> Test version V2.6.4 is available here: https://t.me/AndrOBD_dev/251

Thanks for the update, unfortunately it does not seem to work. Maybe even worse? (The adapter does not return "ELM327 v1.5")
Here are some new HCI Snoop logs if they might help:
[AndrOBD_2.6.4-not_working-with_ATZ.log](https://github.com/fr3ts0n/AndrOBD/files/11072802/AndrOBD_2.6.4-not_working-with_ATZ.log)
[AndrOBD_2.6.4-not_working-without_ATZ.log](https://github.com/fr3ts0n/AndrOBD/files/11072803/AndrOBD_2.6.4-not_working-without_ATZ.log)
[Car_Scanner-working(2).log](https://github.com/fr3ts0n/AndrOBD/files/11072804/Car_Scanner-working.2.log)
[Torque-working.log](https://github.com/fr3ts0n/AndrOBD/files/11072805/Torque-working.log)

I haven't worked with Snoop logs before but from what I can see there seems to be a differences between AndrOBD and CS/TQ. The frame the phone sends right after the adapter responds with "ELM327"  includes ATE0. (Frame 191 in Torque-working.log & 286 in Car_Scanner-working(2).log)
Car Scanner even seems to be sending ATE0 before receiving the "ELM327" response... (Frame 282 in Car_Scanner-working(2).log)
Could this have something to do with it?

fr3ts0n, 2023-03-28:
You can.open hci logs with wireshark for Analysis.We are much earlier than sending ATE command. It seems to happen before the First command is sent.