fr3ts0n/AndrOBD #236: Please help to create ECU SIMULATOR

rakshitbharat, 2023-05-07:
I want to work with this project but I want to create virtual car in my lab, on entire internet there is no proper emulator/ simulator device available.

Please help

Thanks

dumbfingers, 2023-07-03:
There are lots of device available, you can purchase a ready-to-use one or build yourself from Arduino:

https://www.google.co.uk/search?q=obd+ii+simulator&sxsrf=AB5stBhu2oS5b6j1udkDDi-2c4WVSLcfCg%3A1688399327262&ei=3-2iZJGhD6qrhbIPldCd8AQ&ved=0ahUKEwjRjI_R8fL_AhWqVUEAHRVoB04Q4dUDCA8&uact=5&oq=obd+ii+simulator&gs_lcp=Cgxnd3Mtd2l6LXNlcnAQAzIFCAAQgAQyBggAEBYQHjIKCAAQFhAeEA8QCjIICAAQFhAeEAoyBggAEBYQHjIICAAQigUQhgMyCAgAEIoFEIYDMggIABCKBRCGAzoKCAAQRxDWBBCwA0oECEEYAFDLBliFCmCbC2gBcAF4AIABcogBswOSAQM0LjGYAQCgAQHAAQHIAQg&sclient=gws-wiz-serp

hitanshi2996, 2023-08-01:
@dumbfingers 
Thanks for suggestion i have started using below tool for ECU simulator
https://vehelec.com/products/obd-ii-ecu-simulator-iso15765-iso9141-2-iso14230-can-bus
It successfully delivered to India and also working fine.

closed 2023-08-02 by rakshitbharat