fr3ts0n/AndrOBD #237: Manufacturer-specific fault codes

mgord9518, 2023-06-14:
Not a problem, but a small quality-of-life improvement 

Utilize a database of manufacture-specific codes instead of saying to check the manual. 

I think a potential improvement to the current behavior would be to utilize a community-ran database of P-codes, which could be contributed to in order to support manufacturer-specific codes. The database would ideally have its own repo and export to something simple like CSV or JSON, which could be read by AndrOBD.

New behavior could be as follows: search database for any encountered non-standard code, if not found then display a dialog prompting to search the internet for it instead (I know you can already do this but it requires manually closing the dialog then long-pressing the code). The dialog could also include some simple instructions to report the currently unknown code.

fr3ts0n, 2023-06-16:
Thank you for your report.

AndrOBD does have the default behaviour of "Long-Press any DFC" searches the internet for descriptions. This is also for codes which are known to AndrOBD.

The AndrOBD project already does maintain a "database" of known OBD codes. 
This is a (translateable) properties file in library/src/main/java/com/fr3ts0n/ecu/prot/obd/res.

mgord9518, 2023-06-16:
@fr3ts0n How do you recommend to contribute? Eg: if I add a code for a Chevy, is there a possibility that a different manufacturer-specific code from a Ford could have the same ID? I just want to make sure before adding a few so that it doesn't create future problems.

fr3ts0n, 2023-06-16:
Yes, same id may have different meaning for different manufacturers.
That is why AndrOBD only maintains generic DFCs and allows web lookup for all reported codes.
There are quite good OEM communities out there, which provide good explanations.
I prefer to use this knowledge rather than maintaining a multy OEM database. 
The effort for maintenance such a db is quite high... plus without the expertise there is no chance to ensure quality of this data.

closed 2023-08-06 by fr3ts0n