fr3ts0n/AndrOBD #241: BLE support

aha999, 2023-09-03:
**Is your feature request related to a problem? Please describe.**
Currently the AndrOBD app does not support BLE adapters, which then need to operate in BT mode.

**Describe the solution you'd like**
Support for BLE adapters connectivity

**other**
The issue was mentioned in the telegram channel, so i posted it for future reference.

Wave6677, 2024-01-10:
I just bought a device and found out it was BLE because of this post. the F-Droid description should be changed at the very least to indicate this feature isnt available yet :(

fr3ts0n, 2024-01-10:
Is this adapter BLE only?
Could you please provide a link to this adapter?
Thx

derkrasseleo, 2024-04-11:
I tried AndrOBD v2.6.8 with a Carly OBD Adapter and couldn't connect.

helmut72, 2024-04-12:
Same for me. Also using Carly OBD Adapter.

fr3ts0n, 2024-04-16:
Pls could you provide a log file to verify that there is no communications going ...?
Thx

fr3ts0n, 2024-04-20:
Please note that the Carly adapter does NOT claim to be ELM compatible. 
They  advertise  to use an updatable ARM controller.
It appears that the adapter is usable with their software only.
I haven't found any documentation of their protocol used, but it doesn't seem to be ELM compatible.

derkrasseleo, 2024-04-20:
That might be true, but I have used it on android and iOS with "Car Scanner" and other third party apps, which worked..

ieffuie4, 2024-07-05:
I have same issue. Also my adapter is not detected because of missing BLE search functionality in AndrOBD. The people before did not respond to your questions or did not have a universal adapter. I am willing to help adding BLE support with a universal BLE adapter.
I have a OBDLink CX https://www.obdlink.com/products/obdlink-cx/
Its fully functional with other apps that have BLE support build in. Here a example of how the GUI of those other apps is looking like for choosing BLE devices:
[image]
[image]

AndOBD is telling "No devices paired" and not scanning. I expect to be searching in the Android OS for OS-paired devices but this is completely wrong for BLE. If you use the OS to pair a BLE device, then its unusable in the apps. BLE works the opposite. Do NOT pair BLE devices with the OS. Open the App you want to use it with and use the app to connect to the BLE adapter.
I can also make a screenshot if you do not believe me of the OBDLink-app (the manufacturer have an own app to update the firmware of the adapter). In the OBDLink app also the manufacturer is telling to not pair the device with Android-OS but to pair it with the APP itself.

To make the report here even more complete: The firmware version i am running on the adapter is the latest available at the moment. Its Firmware Version 5.10.1 .

fr3ts0n, 2024-07-06:
I am currently looking into BLE driver implementations  ...
Best choice for potential integration so far is this project: (https://github.com/kai-morich/SimpleBluetoothLeTerminal.git)
We are getting there ...

fr3ts0n, 2024-07-11:
Test version with BLE support avaulable now:
https://t.me/AndrOBD_dev/259

derkrasseleo, 2024-07-12:
> Test version with BLE support (V2.7.0) available for testing now: https://t.me/AndrOBD_dev/259

I would love to test it, but I can't download the apk from telegram. Can you maybe link to the file directly?

ieffuie4, 2024-07-12:
Same here. I also cant download the apk because its not available without having to send a mobile phone number to Russia/Emirates. I really like privacy, free speech and human rights.

fr3ts0n, 2024-07-13:
I changed to use the github build action to generate unsigned beta packages for download.
here is the first GitHub generated beta package of V2.7.0 test version for download:
https://github.com/fr3ts0n/AndrOBD/actions/runs/9921537206/artifacts/1698194564

derkrasseleo, 2024-07-15:
I just installed 2.7.0 and I don't see how I can connect a BLE device now. It's only showing me devices that were paired before..

fr3ts0n, 2024-07-15:
You need to select BLE adapter type in OBD settings. After that you will get a selection screen for detected BLE devices on connection attempt.

fr3ts0n, 2024-07-15:
Could you please change the log level to FINE to allow more details in log output. I can not see what UUIDs are served by the adapter.
Thx

derkrasseleo, 2024-07-16:
Here's the log (withouth plugin data logging):

<details><summary>LOG</summary>
<p>

```
2024-07-16	13:25:12.423	INFO	AndrOBD	AndrOBD V2.7.0 starting
2024-07-16	13:25:12.445	FINE	AndrOBD	Adapter: android.bluetooth.BluetoothAdapter@245c880
2024-07-16	13:25:12.575	INFO	AndrOBD	Conversion: metrisch
2
2024-07-16	13:25:17.752	FINE	CommService	connect to: 00:1A:22:01:B4:A7
2024-07-16	13:25:17.753	FINE	CommService	setState() NONE -> CONNECTING
2024-07-16	13:25:17.758	INFO	CommService	BT socket - UUIDs:NONE (Invalid BT implementation)
2024-07-16	13:25:17.760	INFO	CommService	BEGIN mBtConnectThread SocketType:Insecure
2024-07-16	13:25:17.769	FINE	CommService	Connect BT socket

2024-07-16	13:25:22.922	FINE	CommService	read failed, socket might closed or timeout, read ret: -1
2024-07-16	13:25:22.927	INFO	CommService	Closing BT socket
2024-07-16	13:25:22.927	FINE	AndrOBD	PluginData: ACC_Z=6.928418
2024-07-16	13:25:22.934	INFO	CommService	Fallback attempt to create RfComm socket
2024-07-16	13:25:22.947	INFO	CommService	Fallback socket - UUIDs:NONE (Invalid BT implementation)

2024-07-16	13:25:28.117	SEVERE	CommService	read failed, socket might closed or timeout, read ret: -1
2024-07-16	13:25:28.119	FINE	CommService	stop
2024-07-16	13:25:28.118	FINE	AndrOBD	PluginData: ACC_Z=7.8317084
2024-07-16	13:25:28.121	INFO	CommService	Closing BT socket
2024-07-16	13:25:28.125	FINE	CommService	setState() CONNECTING -> OFFLINE
2024-07-16	13:25:28.127	FINE	CommService	setState() OFFLINE -> OFFLINE

2024-07-16	13:25:30.319	FINE	com.fr3ts0n.prot	com.fr3ts0n.ecu.prot.obd.ElmProt@cd0ae2a TX:'ATLP'
2024-07-16	13:25:30.428	FINE	CommService	stop
2024-07-16	13:25:30.430	FINE	CommService	setState() OFFLINE -> OFFLINE
2024-07-16	13:25:30.431	INFO	AndrOBD	AndrOBD V2.7.0 finished
```

</p>
</details>

ieffuie4, 2024-07-16:
I can confirm this issue with the 2.7.0 test release. On the OBDLink CX you have 3 LED's. The blue one is the important one for this test. Its the Bluetooth LED. Its flashing until a connection is made. It never stop flashing when AndrOBD try to connect. All other apps that connect over BLE show that they connected by a not flashing blue LED on the OBDLink CX adapter.
I reconnected the adapter few times. Connection always work with all other OBD2 BLE capable apps, but never with AndrOBD.
The search functionality in AndrOBD is working. I can see the BLE adapter. From the usability point of view i would scan automatic when the window is shown instead of having to press the search icon.

fr3ts0n, 2024-07-17:
Pls could you attach a log file?

See https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues#debug-log-files

Thx

ieffuie4, 2024-07-17:
Sadly i can not. See https://github.com/fr3ts0n/AndrOBD/issues/271
But in the logcat of AndrOBD i can see:
I wm_on_stop_called: [Token=247565260,Component Name=de.kai_morich.simple_bluetooth_le_terminal.MainActivity,Reason=LIFECYCLER_STOP_ACTIVITY,time=1ms]
I wm_on_destroy_called: [Token=247565260,Component Name=de.kai_morich.simple_bluetooth_le_terminal.MainActivity,Reason=performDestroy,time=5ms]
--------- switch to main
W WindowOnBackDispatcher: sendCancelIfRunning: isInProgress=falsecallback=android.view.ViewRootImpl$$ExternalSyntheticLambda11@449327f
I CommService: Closing BT socket
D BluetoothSocket: close() this: PRIVACY-REMOVED-MAC, channel: -1, mSocketIS: android.net.LocalSocketImpl$SocketInputStream@4509b4f, mSocketOS: android.net.LocalSocketImpl$SocketOutputStream@3be62dc, mSocket: android.net.LocalSocket@c574fe5 impl:android.net.LocalSocketImpl@ba2faba fd:java.io.FileDescriptor@2a59d6b, mSocketState: INIT
I CommService: Fallback attempt to create RfComm socket
I CommService: Fallback socket - UUIDs:NONE **(Invalid BT implementation)**
E CommService: read failed, socket might closed or timeout, read ret: -1
I CommService: Closing BT socket

And from the functionality point of view what happens with the other OBD-BLE apps that work: The Android-OS is showing a dropdown window, then showing me the BLE-Pin and when i confirm the BLE connection to the OBD adapter works. This BLE-Pin is never shown with AndrOBD and i do not get any dropdown shown from the Android-OS.

ieffuie4, 2024-07-18:
After i did this https://github.com/fr3ts0n/AndrOBD/issues/271#issuecomment-2236829106
i got now the logfile.

INFO	AndrOBD	AndrOBD V2.7.0 starting
INFO	AndrOBD	Conversion: metric
FINE	CommService	connect to: PRIVACY-REMOVED-MAC
FINE	CommService	setState() NONE -> CONNECTING
INFO	CommService	BT socket - UUIDs:NONE (Invalid BT implementation)
INFO	CommService	BEGIN mBtConnectThread SocketType:Secure
FINE	CommService	Connect BT socket
FINE	CommService	read failed, socket might closed or timeout, read ret: -1
INFO	CommService	Closing BT socket
INFO	CommService	Fallback attempt to create RfComm socket
INFO	CommService	Fallback socket - UUIDs:NONE (Invalid BT implementation)
SEVERE	CommService	read failed, socket might closed or timeout, read ret: -1
FINE	CommService	stop
INFO	CommService	Closing BT socket
FINE	CommService	setState() CONNECTING -> OFFLINE
FINE	CommService	setState() OFFLINE -> OFFLINE

ieffuie4, 2026-01-08:
Any news about BLE support would be nice.

fr3ts0n, 2026-02-20:
Beta version (V2.7.0) with BLE support is here: https://github.com/fr3ts0n/AndrOBD/actions/runs/22216484272/artifacts/5586219788
I've tested with 2 adapters so far:
- VGate iCar Pro
- Nexas NexLink

Please test with your BLE adapters and confirm operation, or report with (fine) logfiles ...