fr3ts0n/AndrOBD #243: bug: "Load measurement": Crash when loading file picker

xnumad, 2023-10-13:
# Describe the bug

App closes when clicking "Load measurement" in menu.
If I try this multiple times, I get "Media keeps stopping" Android app crash dialog.

## To Reproduce

Steps to reproduce the behavior:
1. Open menu
2. Tap "Load measurement"
3. Observe application crash (going away without animation back to app launcher)

## Expected behavior

A file picker opens, to be able to select a saved measurement.

## AndrOBD Debug log files

Even with log level ALL, the log doesn't contain anything about the error.

## Important details on your environment:

- Android OS version: 13
- AndrOBD version: 2.6.6
- OBD adapter type: N/A
- Connected vehicle make, type, year, engine size: N/A

## Additional context

from `adb logcat`:
```
10-12 11:23:14.603 13944 13944 I PhotoPickerActivity: Launch DocumentsUI and finish picker
10-12 11:23:14.604 13944 13944 D AndroidRuntime: Shutting down VM
10-12 11:23:14.604 13944 13944 E AndroidRuntime: FATAL EXCEPTION: main
10-12 11:23:14.604 13944 13944 E AndroidRuntime: Process: com.google.android.providers.media.module:PhotoPicker, PID: 13944
10-12 11:23:14.604 13944 13944 E AndroidRuntime: java.lang.RuntimeException: Unable to start activity ComponentInfo{com.google.android.providers.media.module/com.android.providers.media.photopicker.PhotoPickerActivity}: android.os.FileUriExposedException: file:///storage/emulated/0/com.fr3ts0n.ecu.gui.androbd exposed beyond app through Intent.getData()
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3644)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3781)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:101)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:138)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2306)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:106)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.os.Looper.loopOnce(Looper.java:201)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:288)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:7918)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
10-12 11:23:14.604 13944 13944 E AndroidRuntime: Caused by: android.os.FileUriExposedException: file:///storage/emulated/0/com.fr3ts0n.ecu.gui.androbd exposed beyond app through Intent.getData()
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.os.StrictMode.onFileUriExposed(StrictMode.java:2208)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.net.Uri.checkFileUriExposed(Uri.java:2407)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.content.Intent.prepareToLeaveProcess(Intent.java:11914)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.content.Intent.prepareToLeaveProcess(Intent.java:11863)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.Instrumentation.execStartActivity(Instrumentation.java:2051)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.Activity.startActivityAsUser(Activity.java:5742)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.Activity.startActivityAsUser(Activity.java:5717)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at com.android.providers.media.photopicker.PhotoPickerActivity.launchDocumentsUiAndFinishPicker(PhotoPickerActivity.java:355)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at com.android.providers.media.photopicker.PhotoPickerActivity.rerouteGetContentRequestIfRequired(PhotoPickerActivity.java:341)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at com.android.providers.media.photopicker.PhotoPickerActivity.onCreate(PhotoPickerActivity.java:134)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.Activity.performCreate(Activity.java:8342)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.Activity.performCreate(Activity.java:8321)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1417)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3625)
10-12 11:23:14.604 13944 13944 E AndroidRuntime:        ... 12 more
10-12 11:23:14.610  1707  5672 W ActivityTaskManager:   Force finishing activity com.google.android.providers.media.module/com.android.providers.media.photopicker.PhotoPickerGetContentActivity
10-12 11:23:14.622  1707  5672 W ActivityTaskManager:   Force finishing activity com.fr3ts0n.ecu.gui.androbd/.MainActivity
10-12 11:23:14.625 13944 13944 I Process : Sending signal. PID: 13944 SIG: 9
10-12 11:23:14.634  1096  1096 I Zygote  : Process 13944 exited due to signal 9 (Killed)
10-12 11:23:14.635  1707  5672 I ActivityManager: Process com.google.android.providers.media.module:PhotoPicker (pid 13944) has died: fg  TOP 
10-12 11:23:14.635  4231  4282 D BinderSender: onProcessDied: pid=13944, uid=10219
10-12 11:23:14.639  1707  2131 I libprocessgroup: Successfully killed process cgroup uid 10219 pid 13944 in 0ms
```

fr3ts0n, 2023-11-03:
I have re-checked into this problem on a Android 13 device using the un-changed AndrOBD release:
- Any attempt of re-loading any measurement file does correctly get served by the default file manager
- Measurement files can be selected, loaded and viewed / analyzed cleanly

The error message above (logcat) does indicate a problem in your used photo picker app. That is the app, which crashes based on the file browse request of AndrOBD. 
- The cause of your reported problem appears to be a problem in your Android / app environment rather than AndrOBD.

I will also abort the suggested PR, since:
- the problem is not caused by AndrOBD 
- the proposed change would break backwards compatibility
- the proposed change would increase app size by 50%

BaderSZ, 2024-03-23:
Not to re-ping an old issue, but I'd prefer to continue here aside from opening a new one:

So far, I've tried to load/open measurements on two different devices (Samsung with Android 13, Pixel 6a with 14) and both of them end up crashing the app. The only way I can load or open measurements is by installing the app in android studio's emulator or Android-x86, which is (and I don't want to seem rude) a ridiculous workaround.

Could you suggest a better solution or workaround for users with this issue to be able to load the file? Or are we really worried about adding 700KB to a 1.4MB app and keeping support for decade-old, unpatched versions of Android?

fr3ts0n, 2024-03-27:
I have to agree, that this change is definitely needed for support Android 14. My Android 13 devices all seem to be happy without it .
I really hate the way Android behaviour and handling changes. 
Android somehow appears to degenerate to a huge collection of restricions, which all require workarounds ... ;-(

However, I'll implement your proposed changes :+1:

kyrlon, 2024-05-05:
> Test version V2.6.9: https://t.me/AndrOBD_dev/256

Came here with the Load measurement problem on my Galaxy S21+ running Android 14, and with this version I was able to load my saved measurement finally!! Thanks!

closed 2024-05-05 by fr3ts0n