fr3ts0n/AndrOBD #246: AndrOBD is not reading some specific  DTC's

DevEnthusiast99, 2023-11-02:
# Describe the bug

On Toyota Sai 2011, Nissan Note 2012 vehicles, App is not finding DTCs even though there are multiple MIL lights are ON in the dashboard.

## To Reproduce

Steps to reproduce the behavior:
1. Try on the above mentioned models or any models 

## Expected behavior

When I tried the same cars with Carscanner app, everything works and I was able to clear the DTCs

## AndrOBD Debug log files

Not available

## Important details on your environment:

- Android OS version: 15
- AndrOBD version: 2.6.6
- OBD adapter type: ELM327
- Connected vehicle make, type, year, engine size: Toyota Sai 2011, Nissan Note 2012

## Additional context
Picture from the Dashboard of Toyota Sai 2011 below:

[image]

DevEnthusiast99, 2023-11-03:
Sure. I will try to provide a log file. I will definitely follow the given instruction. Thx.

DevEnthusiast99, 2023-11-03:
Hi again, 
I haven't been lucky about getting the Logs as asked. 
1. I set the Log level to ALL but  I found only empty file in the Logs directory unfortunately. 
2. I also tried things like disabling ATZ command, and Enabling Reset on NRC. 

Since I could not get access to the same SAI described in my original issue. I tested on 2008 Toyota RAV4 but I got "Protocol not supported error" every time I try OBD data and OBD fault codes in the app. 

Any input on this issue can help a lot :) 

By the way, is the current code supporting cars before 2008? Because I've haven't been successful with cars older than 2008 due to the same "Protocol not supported error"

[image]

fr3ts0n, 2023-11-03:
A 2008 RAV4 should definitely support OBD data and DFC services.
It is really impossible to analyze anything protocol related without any log output ...

Have you checked the app file permissions to be allowed to write log files?
- App info - Permissions