fr3ts0n/AndrOBD #248: 关于私有协议

276016439, 2023-12-11:
请问一下，已知canid的情况下，是否可以请求返回此id的数据吗？

fr3ts0n, 2023-12-11:
Is there any chance you yould ask this in english?
Thx.

fr3ts0n, 2023-12-12:
If I understood your question correctly you ask if a OBD request can be sent directly to a known CAN partner / CAN id. 

The answer is yes, but... ;)
Yes, you can talk OBD directly on CAN using OBD service ids in UDS based message frames.
I.e.: use SVC 0x01 (OBD data) instead 0x22 (UDS RDBI)
Your OBD request will only be answered by an ECU which services OBD protocol. Commonly this is only Engine control ECUs, since these are  relevant for exhaust emissions.
Wikipedia is a good resource to learn about these protocols.

276016439, 2023-12-12:
Thank you very much for your reply. I am a beginner and have just started working with EML327. My requirement is to send instructions to OBD through EML327. Can OBD return data such as canid=388?

fr3ts0n, 2023-12-15:
Yes, CANID is part of OBD communication if it's communicated over CAN lines (ISO 15765)  

I really would like to suggest you to read about vehicle communication protocols:
- https://en.wikipedia.org/wiki/On-board_diagnostics
- https://www.elmelectronics.com/DSheets/ELM327DSL.pdf

276016439, 2023-12-15:
> Yes, CANID is part of OBD communication if it's communicated over CAN lines (ISO 15765)  
> 
> I really would like to suggest you to read about vehicle communication protocols:
> - https://en.wikipedia.org/wiki/On-board_diagnostics
> - https://www.elmelectronics.com/DSheets/ELM327DSL.pdf

Thank you very much！！！

closed 2023-12-20 by fr3ts0n

machoaga, 2024-02-03:
> If I understood your question correctly you ask if a OBD request can be sent directly to a known CAN partner / CAN id.
>
> The answer is yes, but... ;) Yes, you can talk OBD directly on CAN using OBD [...]

i've managed to talk to another ECU of the car (the Headunit/Navigation Display) using its CAN ID for request and response as AT SH and ATCRA respectively. its kinda successful as i can send UDS 11 01 for a Hard Reset and it restarts the Headunit. However the ELM327 doesnt like the answer, so always shows NO DATA as a response, anyone got any pointers as in what could be misconfigured ?

fr3ts0n, 2024-02-08:
11 01 is a UDS service, but not a OBD one.
This service is a questionable example, since it performs a reset which potentially does not generate a UDS response.
There may be other common UDS services (like RDBI (22 xxxx)), which should be more consistent in responding.
But as I mentioned above. It's not generic OBD, it's OEM specific UDS protocol.