fr3ts0n/AndrOBD #250: Drivers log

IzzySoft, 2023-12-12:
**Is your feature request related to a problem? Please describe.**
Many data required for tax purposes are available automatically via ODB2 (e.g. mileage at start/end of a trip, fuel consumption, locations). It would be great if they could be utilized to automate collecting them for that purpose.

**Describe the solution you'd like**
Collected data could be used to pre-fill a drivers log. See this photo of a German paper book for this purpose:

[image]

Columns translated:

* date
* time (from/to aka trip-start/end)
* route/target, purpose
* purpose / visited companies, driver / occupants / cargo
* mileage at start
* mileage at end
* driven kilometers (business, private, home/work)
* fuel (litres, cost)
* oil (litres, cost)
* others (cost, purpose)

Many of these fields could be pre-filled via ODB2-collected data. Some others could have often used values prepared (like often visited companies/locations or routes/targets). Not all of them are even mandatory – but have as much data as possible gathered automatically would be a tremendous help already – especially against forgetting to write them into some entirely manual "solution" like a paper book.

**Describe alternatives you've considered**
A paper logbook (inconvenient, one tends to forget). A separate app (found no FOSS app for it).

**Additional context**
This would be a nice complement to using the app as dashboard. Thanks in advance for considering!

fr3ts0n, 2023-12-20:
Hello Izzy,

Thanks for your idea.

The OBD regulation just mandates vehicle manufacturers to report data which is relevant for vehicle exhaust emissions.
In your list of fields I am afraid I do only find some fields which may be sourced from OBD data:
- Mileage
and the derivates
- driven kilometers

However, there is a OBD PID reserved for absolute vehicle mileage, but unfortunately this data field is NOT mandatory for OBD data output and therefore many vehicles do NOT support it for OBD data requests. 
On the selection of three vehicles which I can get my hands on frequently, neither of them does provide it's mileage on OBD request. There are several other data items like "Miles driven with MIL active", "Miles driven since last reset", etc. but neither of these does represent the absolute mileage.

On items like Fuel and Oil, (including prizes) I assume this would need to be hand-entered, since neither fuel level, nor oil level are metered/reported via OBD protocol.

Many of this data potentially is available in OEM specific diagnostic protocols, which are used in OEM specific test solutions.
The effort to support these protocols for the most common vehicles does definitely exceed the level of a free OSS project.

I see that there are some solutions available to keep a semi-automated drivers log, but I understand that these are payment apps, most likely for above reason...

IzzySoft, 2023-12-20:
Thanks for your reply, @fr3ts0n! I'm just beginning to play with OBD, so my experiences are still close to zero (my last car 30 years ago didn't have OBD or I didn't know of it back then, and I just started to look into it when I got a new car now), so I don't know how much I can help with the details.

> On the selection of three vehicles which I can get my hands on frequently, neither of them does provide it's mileage on OBD request.

If you tell me how to do that, I could check with mine to possibly add a forth (a Skoda Fabia). I'm using "that Dongle" btw (vgate iCar Pro – found some issue related to it here afterwards, hence "that").

> "Miles driven with MIL active"

That field always showed 0 here, and I have not even a clue what MIL stands for :see_no_evil: Maybe it's something I could enable – but not even knowing what it is I cannot even check…

> On items like Fuel and Oil, (including prizes) I assume this would need to be hand-entered, since neither fuel level, nor oil level are metered/reported via OBD protocol.

Yes, I've assumed it for those. Especially for Oil that's no big deal as it usually does not happen *that* frequent. And for fuel there's always the receipt reminding you (I still need to find out if I'll have to enter the full amount of fuel there, or just what matches the corresponding trip). I was hoping for (GPS) Coordinates, though.

> The effort to support these protocols for the most common vehicles does definitely exceed the level of a free OSS project.

Sure – and I didn't expect you adding vendor-specific stuff (your project description is clear on that, and I fully understand it and agree). Maybe those interested *and* having the knowledge+time could maybe offer such via plugins, but as long as no such plugin exists it's out of question for *this* issue here.

So with the facts at hand (and possibly results from my test if I can provide such), would you consider having a "drivers log" feature included with AndrODB in general? The pros compared to fully manual logs is that at least trips could be recorded automatically, with their dates/times and maybe even with the kilometers pre-filled. Makes it at least easier to "catch up" when one forgot (or if there are multiple drivers, to find out "who the heck was driving those 250km last week, when and where to" as one had something to "hold on".

> I see that there are some solutions available to keep a semi-automated drivers log, but I understand that these are payment apps, most likely for above reason...

Not only that. Most (if not all) of such solutions I found were bound to some online service, so all your data would be on some server you'd need to trust (apart from the fact they are already on the manufacturers' servers which you cannot control (in my case, "Skoda Connect") – in an "eat-or-die" way (you cannot use e.g. traffic control without having that service enabled)). And in the FOSS world live many of us who don't like spreading their personal data across "unknown services"…

IzzySoft, 2023-12-20:
PS: [here](https://www.kassel-steuer.de/service/artikel/fahrtenbuch/) are some details on the needed values, page is by a tax consultancy (and in German). An example log is in the middle of the page. So the first two columns (date & time) could come from OBD, the 3rd column (if GPS can be taken from ODB) could be retrieved via a Nominatim service on-device (AFAIK most Android devices have that). Mileage as discussed above. The forth and the last 6 columns would be manual (or the driven kilometers could be placed e.g. into "geschäftlich" and left for manual transfer/splitting).

And yes, due to the fact one can edit entries, this won't be sufficient for the tax office (see the corresponding hint on the linked page, items 4 & 5; nobody can expect you getting the app certified for that¹). But it would definitely be helpful in avoiding "gaps" and having to ponder about what might be missing.

----

¹ though going by [this page](https://vimcar.de/fahrtenbuch/rechtssicherheit), according to "[BFH-Urteil vom 16.3.2006, BStBl. II, S. 625](https://datenbank.nwb.de/Dokument/195006/)" it must only be ensured that date, time and mileage of a trip cannot be "edited". So if you'd be interested in "making it proof", some details [can be found here](https://www.gpsfahrzeugortung.de/hilfe/elektronisches-fahrtenbuch-finanzamt.html) and, more importand with the corresponding hints [here](https://www.kfz-fahrtenbuch.de/vorgaben-finanzamt/) (all in German – but as you're from Austria I guess that's not a problem) – and I could ask my tax consultant if they can get an "approval statement" from the tax authority (concerning Germany, but you might be able to use that for Austria as well when presenting it to an authority there, to ease that part for your country).

fr3ts0n, 2023-12-20:
The test for checking the availability of Mileage data is rather easy:
- Connect to vehicle
- Open OBD Data screen
- Check if OBD data list contains following data item
  - Odometer (english)
  - KM-Stand (german)

If it is contained in the data list you are lucky.

> "Miles driven with MIL active"

It is a good sign for your car, if this item shows 0. 👍 
MIL stands for Malfunction Indicator Lamp, which is the OBD Check engine light
So if this counter counts up this means you are ignoring / have ignored a failure condition.

IzzySoft, 2023-12-27:
Apologies for the delayed response, was visiting family over the holidays. Took that as chance to check the data, though. Unfortunately, could not find either Odometer nor KM-Stand. So it seems we#re out of luck there :cry: If you can tell me how to perform a full CSV export of all fields for a trip, that would probably make a search easier for what I might have missed. No idea even how to screenshot the full list…

Here's what I found with "km":

* distance since ECU reset: 0
* distance since MIL activated: 0
* avg distance between DPF regenerations: 0.00

Some fields are marked "Reserved":

* Boost Control status A
* Boost Control status B

There's also a bunch of fields with "() available" and "() incomplete", only few have one of the fields checked with an asterisk.

I could of course provide you with a full `*.obd` file if you'd want one, but would prefer to not attach it here :wink: 

> MIL stands for Malfunction Indicator Lamp

Oh – yeah, that would be a bad sign for a brand new car indeed :see_no_evil: