fr3ts0n/AndrOBD #251: Bug report

jtzf520, 2023-12-15:
## Expected behavior

Press any buttons to check vehicle info, vehicle data etc... wont work

## AndrOBD Debug log files

[image]

### Log detail level

ALL

## Important details on your environment:

- Android OS version: Android 13
- AndrOBD version: v2.6.7
- OBD adapter type: Type A
- Connected vehicle make, type, year, engine size: Perodua BEZZA, Sedan, 2016, 1.3l

## Additional context

Add any other context about the problem here.

I tried to link the device through bluetooth but nothing appeared in the screen whenever i tried to press vehicle info or any other stuff.

fr3ts0n, 2023-12-15:
In your log file I cannot detect any communication from/to your BT adapter.

The phone / app does not appear to get any confirmation of the BT connection attempt. 
It re-tries connection with a fallback socket, but no confirmation / communication appears there as well :cry: 

```
Communication path:
[phone] ---x--- [BT adapter] ---?--- [ECU]
```

Pls can you assure for you test:
- BT adapter is connected to your vehicle
- BT adapter is powered (LEDs are on / blinking)
- BT adapter is detected and coupled to the phone correctly

I can see in the log that you try to connect with a secured BT socket. 
I know of some adaptors in the field which fail on secured BT socket connection. (however, that failure did show differently in the logs).
It still could be worth a try to disable the secure BT connection in the app settings.

Thx

jtzf520, 2023-12-15:
Hi...it's solved already, thank you very much for your help :)

closed 2023-12-15 by jtzf520

fr3ts0n, 2023-12-15:
Just out of curiosity: What was the cause?

jtzf520, 2023-12-20:
Sorry for the late respond, was busy these few days...btw, its the secured bt connection tht you mentioned. I've disabled it and it works just fine :)

fr3ts0n, 2023-12-20:
Would it be possible for you to post your experience with your adapter to following query issue?
#187 

This might help others as well ...
Thx