fr3ts0n/AndrOBD #252: "I have a question about sending a portion of the OBD data to my server."

ham8962, 2023-12-25:
"I would like to express my gratitude for releasing such a great application for free, along with its source code. I am a university student working on my personal project based on this application, and I am writing this to ask a question about the code.

What I want to do is to send a portion of the vehicle's data to the server I built myself (I am currently planning to send only the speed, RPM, and Odometer). In order to capture the data I intend to send to the server, I read the OBDGaugeAdapter Class and OBDitemAdapter Class, as well as the files under the related library package (I mean the files existing in the ecu directory), and understood a bit of the flow. However, I haven't reached the level of understanding to edit the code.

I would like to get help on how the flow of the code proceeds and which classes I should look at to achieve my goal (I would also like to know how to capture some of the OBD data).

Once again, thank you for releasing this great application as open source on Github."

fr3ts0n, 2024-01-02:
This sounds like another plugin extension to me ...
Please take a look on the MQTT extension which does very similar things with a MQTT broker
If you replace the data sending with your server protocol you should get rather close ...

ham8962, 2024-09-28:
안녕하세요 오랜만입니다. 다른 분야의 공부를 하다 기존 프로젝트를 발전시키고 싶어 연락드렸습니다. 

mqtt와 gps 기능을 사용하고 싶어 andrOBD-plugin 을 기존 androOBD 프로그램의 Plugin 패키지에 위치 시켰습니다 (일단 여기까지만 진행했습니다. - 혹시 뭔가 잘못한걸까요?)

에러가 좀 뜨는데 라이브러리쪽 문제인거 같습니다. 이미지를 보시고 혹시 무엇이 문제인지 가르쳐주실 수 있으신가요?
(더 필요하신 점이 있으시면 말해주세요 바로 업데이트 하겠습니다.)  

I see some errors, but I think it's a library problem. Can you look at the image and tell me what the problem is?
[image]
[image]