fr3ts0n/AndrOBD #257: OBD protocol error (NRC:0x11) Service 0x0A not supported

rustysixslinger, 2024-03-14:
# Describe the bug

**Receive the error message of "OBD protocol error (NRC:0x11) Service 0x0A not supported" when trying to view and clear DTCs. I am unable to view DTCs. When trying to clear DTCs, error message is shown but DTCs are cleared when the error message is closed.**

## To Reproduce

Steps to reproduce the behavior:
**Attempt to view or clear DTCs.**

## AndrOBD Debug log files

**Log file with FINE logging option attached**
[AndrOBD.log.1.txt](https://github.com/fr3ts0n/AndrOBD/files/14595926/AndrOBD.log.1.txt)
.

## Important details on your environment:

- Android OS version: **14**
- AndrOBD version: **2.6.7**
- OBD adapter type: **ELM 327 MINI (bluetooth)**  
[image]
- Connected vehicle make, type, year, engine size: **Toyota, 4Runner, 2008, V6**

## Additional context

**I AM able to view live OBD data without error or issue which makes this issue odd. 

Issue is present AndrOBD v2.6.7, v2.6.6 and v2.6.5 in Android version 13 and 14. Issue is not present, and all works as should, in v2.6.4 in Android version 13 or 14.**

rustysixslinger, 2024-03-14:
On doing some further research on this, it appears that the response of 7F0A11 is being sent back because my ECU does not support PDTCs. PDTCs first started to be introduced in 2010 as a slow 3-year rollout to then have them to be manditory.

Just spitballing, but is it possible to catch this response, ignore and move on or possibly not perform PTDC checks if data is received indicating that the ECU is pre 2010?

SpookySkeletons, 2024-03-14:
Is there a quick way in source to zap this check out?

rustysixslinger, 2024-03-16:
> Is there a quick way in source to zap this check out?

Start with looking at all references to OBD_SVC_PERMACODES

rustysixslinger, 2024-03-16:
Log file attached that shows that there are to 2 fault codes. Attempted to view twice so that will show in the log file. Fault codes are not displayed after clicking away from the error message in the attached screenshot. Fault codes were cleared, while receiving the same error message, and this is shown in the log file as well as indicated by the MIL light on the dash going off.

[AndrOBD.log.0.txt](https://github.com/fr3ts0n/AndrOBD/files/14624571/AndrOBD.log.0.txt)

[image]

rustysixslinger, 2024-03-16:
Last addition to this issue unless anything is requested. I'm not a developer and throwing a new protocol in the mix is a bit beyond me but I can say that I'm 100% sure the issue initially reported is due to permanent DTCs not being supported by my ECU. 

[image]

fr3ts0n, 2024-03-17:
Ok, I understand that you car does not support OBD service 0x0A (Permanent Fault codes), which it correctly responds with NRC 0x11 (Service not supported).
So the app does clear the DFC's, but when reading them (to verify clearance) it gets annswer by the car that it doesn't support mode 0x0A.

The problem does occur after AndrOBD change to combine all 3 versions of fault codes (OBD modes 0x03 (DFC), 0x07 (PENDING DFC), 0x0A (PERMANENT DFC))

Since the response if NRC 11 usually is a serious problem for OBD diagnostics, I'll have to find a way to silently accept it in this case ...

fr3ts0n, 2024-03-20:
Test version is availabe here: 
https://t.me/c/1434285001/2318

Please verify if the fix matches your needs

rustysixslinger, 2024-03-24:
I don't have a telegram account and can't access/download the test/beta version without one. I'll attempt to sign up to the service and report back if I can get it downloaded, installed and tested.

rustysixslinger, 2024-07-25:
The latest release, and I thank you for publishing it, still did not solve the issue stated. From what I've seen, suppression of the error was stopped but continuation of checking for DTCs and reporting them back was also skipped.

Can I give you at least a small recommendation? Provide an option in the application settings (enabled by default) to disable checking for perm DTCs. That's it. Period.

I'll send you my code to get this to work if you'd like but that's all I really did. Disable perm DTC checks.