fr3ts0n/AndrOBD #258: Samsung warns AndrOBD wakes often, and wants me to deep sleep it.

OneOfMany07, 2024-03-18:
# Describe the bug

Without actively using the application (haven't ran it for days) I'm seeing warnings about excessive "wakes" from the built in Battery features of my Samsung A71 phone (SM-A716V).  I did try to use your app for the first time a few days ago, and I tried my best to close out of it fully at the end.

I'm sure I could work around the issue, or use the suggested setting (which would prevent the app from running unless I view it), but I thought I'd pass along the message in case it's something you can fix.

My understanding is the phone can't go into deep sleep (saving lots of battery when idle) if ANY application wakes it.  Sure, if I was actively using something in AndrOBD I'd want it to wake, but I haven't run it for days now and never did see data come out of my adapter with it (Torque Lite worked though).

## To Reproduce

1. I installed 3 parts from F-prot (including GPS and Sensor providers, didn't know if needed MQTTPublisher)
2. Ran them and tried to connect to my older bluetooth only adapter (Kobra brand)
3. Didn't get it to work so I tried to close it (swiped away on running app list)
4. Waited a day or two and I periodically see a notification with...

"Extend your battery life
AndrOBD is waking up frequently and using too much battery.  Put it in deep sleep to extend your battery life."

## Expected behavior

No warnings, even when using default "Optimized" mode for the app (versus needing me to manually pick either "Unrestricted" or "Restricted" to prevent these today, and maybe lose battery life or app functionality)

## Important details on your environment:

- (Samsung) One UI version: 5.1
- Android OS version: 13
- AndrOBD version: v2.6.7
- Adapter: older bluetooth only Kobra (like 5-10 years old?)

fr3ts0n, 2024-03-20:
Thanks foryour detailled report.
There should not be any system event waking up AndrOBD main app, except if there is any plugin, which keeps sending data in the background. These plugin activities are typically also stopped on regular closing the AndrOBD app.

Did you ever try to do that test without any plugin activated?
You can de-activate these in the plugin-manager of AndrOBD app.

Thx

OneOfMany07, 2024-03-20:
Sounds like the plugin architecture needs to change.  That the plugins should wait for a "start" message from the main app, and sleep otherwise.  Though I've never tried to design one myself, and don't know about Android app limitations (I'm sure there are many).

I just disabled the 2 plugins from the plugin manager, and will wait to see if I get any other messages.

If it's the plugin(s) causing the issue, then shouldn't the message be about the plugin app instead?  To change their behavior, versus the main app's.

I haven't seen another message today, so I almost wonder if I hadn't closed the app out correctly before.  And the issue is more about "app that shouldn't be running or doing anything was".  That it's not quite sleeping as it could/should.

Or maybe it won't warn me again since it already did, and this isn't a 'new' problem to the phone.  I don't know much about that battery optimization feature, beyond being told to disable it for the particular app I'd first saw this for (to let it run).  I'll look into it more later.

smith901, 2024-03-27:
Noticed the same behavior. In Settings->Development Options->Logging Level has to be set to minimum level of `FINEST` for the app to constantly use power.     

As far as I can tell, logging is turned on here [at line 571](https://github.com/fr3ts0n/AndrOBD/blob/master/androbd/src/main/java/com/fr3ts0n/ecu/gui/androbd/MainActivity.java#L571) but never turned off when [app is closed](https://github.com/fr3ts0n/AndrOBD/blob/master/androbd/src/main/java/com/fr3ts0n/ecu/gui/androbd/MainActivity.java#L710). Current log file at `/storage/emulated/0/com.fr3ts0n.ecu.gui.androbd/log/` constantly grows after app is closed. Doesn't stop growing until app is forced stopped.

Sample of repeated lines after app is closed but NOT forced stopped:    
`2024-03-26	21:56:24.040	FINEST	AndrOBD	Handler notification: { when=0 what=7 target=com.fr3ts0n.ecu.gui.androbd.MainActivity$3 }
2024-03-26	21:56:24.290	FINEST	AndrOBD	Handler notification: { when=-1ms what=7 target=com.fr3ts0n.ecu.gui.androbd.MainActivity$3 }
2024-03-26	21:56:24.540	FINEST	AndrOBD	Handler notification: { when=0 what=7 target=com.fr3ts0n.ecu.gui.androbd.MainActivity$3 }
2024-03-26	21:56:24.790	FINEST	AndrOBD	Handler notification: { when=0 what=7 target=com.fr3ts0n.ecu.gui.androbd.MainActivity$3 }
2024-03-26	21:56:25.040	FINEST	AndrOBD	Handler notification: { when=0 what=7 target=com.fr3ts0n.ecu.gui.androbd.MainActivity$3 }
2024-03-26	21:56:25.291	FINEST	AndrOBD	Handler notification: { when=-1ms what=7 target=com.fr3ts0n.ecu.gui.androbd.MainActivity$3 }`

fr3ts0n, 2024-03-31:
The event you see here is the event of screen refresh timer.
This timer is still running when the app is paused. It will stop when the app is finished with double return or force-stopped. 
I'll change that, since there is no need in doing cyclic screen updates on a non-visible app in background.

However, just to make you aware: 
AndrOBD is designed to perform background operations like OBD measurement recording on a road trip. These Actions will also continue to run with the app hidden in background. So if you put AndrOBD to background while in data display mode, it still will continue to read (and record) OBD data. Needless to say that this background activity (if used) needs some energy.

fr3ts0n, 2024-04-01:
Test version (V2.6.10) with corresponding fixes is available here: https://t.me/AndrOBD_dev/257