fr3ts0n/AndrOBD #263: Suggestion for Blutooth OBDII device.

EvanCarroll, 2024-05-11:
The docs and issue tracker frequently say things like,

> Many of these Bluetooth Mini adapters are known to work not too well on ISO9141 systems which your vehicle indicates.

It would be nice if there was a _suggested_ or known-to-work Bluetooth OBDII adapter. Certainly, there must be one that is worth pushing. 

Otherwise, we'll all just buy expensive ones that may not work.

fr3ts0n, 2024-05-12:
Sorry, but it's really hard to give any recommendation, since the market constantly changes.

If you get the identical device twice from the same dealer  within 6 months, you were quite lucky. 

That's why I started to ask the users to give their feedback here: #187 
- This should at least indicate some trends what kind of adapter is usable for what kind of OBD system.

EvanCarroll, 2024-05-13:
What about the name brand ones, like the one by Blue. I realize this is crazy expense, and that's because you're paying for their software. But will that work with it?

ieffuie4, 2024-07-05:
Take just the known working one: OBDLink LX. If you want to spend more money, then you also can go for OBDLink MX+. The additional functionality of the OBDLink MX+ is not really supported by this AndrOBD app. So saving some money and getting the LX is fine.

Note: The newest OBDLink CX with BLE version 5.1 is currently unsupported by AndrOBD. If you do not want to commit to software development now, then do not take the CX but either LX or the MX+.