fr3ts0n/AndrOBD #270: Asking for too much permissions

ieffuie4, 2024-07-17:
# Describe the bug

Recent AndrOBD version ask for way too much permissions with no need.

[image]

It sort of ask for "give me access to everything". Pictures, music, full storage.
I also do not understand why it ask for Network access. I looked into the different options but was not able to find any functionality that is internet or network based. Is it for the MQTTPublisher addon?

I spend some minutes on a walk thinking what would be best example to show how the permissions should look like. I think i found the two best examples:

First and general: 
Take a look into the permissions implementations of LibreTube. They work great for a full app with internet access that is downloading things from the internet into the internal memory of the phone.

Second:
Take a look into the permissions of OsmAnd. There also (huge) downloads are happening and also addons are available. But the app does not ask you for access to simply the whole memory of the phone like AndrOBD does at the moment.

## To Reproduce

Steps to reproduce the behavior:
1. Open AndrOBD and accepp the permissions
2. Go to Settings of your phone and look into the permissions AndrOBD asked for

## Expected behavior

The minimal permissions that are required to my knowledge for a OBD-Bluetooth-App without internet access are:
"Nerby devices" (for Bluetooth)
"Sensors" (also somehow Bluetooth related to my understanding)

Some additional that are optional and could make sense for some usecases:
"Network" (maybe for MQTT?)
"Notifications" (maybe for some other things?)

Those should never be asked for:
Files, Music and audio, Photos and videos

## Important details on your environment:

- Android OS version: Android 14
- AndrOBD version: 2.7.0 and 2.6.12
- OBD adapter type: not related
- Connected vehicle make, type, year, engine size: not related

fr3ts0n, 2024-07-18:
All these permissions are required by AndrOBD, since the permission handling of AndrOID did change a lot. 
If your app is intended to work on more than the latest version of Android, this complete mess of permissions is required!
(sorry for that, but please comment to Android developers ...)
Files / Music + Audio / Photos + Videos
- needed for file access to
  - Wite log file
  - Read / Write measurement files
  - Export CSV fikes
Nearby devices
- needed for Bluetooth access
Network
- needed to handle WiFi adapters
Sensors
- Needed to handle background operations (data recording with disabled display)

All permissions are listed, and commented in the manifest file within the project.

ieffuie4, 2024-07-18:
> If your app is intended to work on more than the latest version of Android, this complete mess of permissions is required!

Could you take a look in the sourcecode of https://github.com/libre-tube/LibreTube ? Its Android 5.0+ and do not require any of such deep permissions like AndrOBD do at the moment.

ieffuie4, 2024-07-18:
After looking around some more, it would make sense to bump the minimum Android version to 5.0.
Take a look here:

https://punchthrough.com/android-ble-guide/

Quote: 
_BLE in the Android SDK

Note: As explained in our[ Android BLE Development Tips](https://punchthrough.com/android-ble-development-tips/) article, we assume the app targets a **minimum of API 21 (Android 5.0)** due to the availability of better BLE APIs such as BluetoothLeScanner and ScanFilter.

We start this section by introducing the main classes from the Android SDK we’ll be using._