fr3ts0n/AndrOBD #271: AndrOBD does not create any logfiles or any other files

ieffuie4, 2024-07-17:
# Describe the bug

I would like to take a look into the logfiles of AndrOBD like requested here: https://github.com/fr3ts0n/AndrOBD/issues/241#issuecomment-2232391588

Sadly this is not possible. AndrOBD seem to have issues with permissions, file write access and so on.
Happily i am able to access the Android logfiles (logcat) related to every single App installed.

I filtered to "error" to make the size better to handle and here is the output:

packageName: com.fr3ts0n.ecu.gui.androbd:20700
buffers: main,system,crash,events,kernel
level: error
--------- beginning of main
 E AndrOBD : /storage/emulated/0/com.fr3ts0n.ecu.gui.androbd/log
 E AndrOBD : java.nio.file.NoSuchFileException: /storage/emulated/0/com.fr3ts0n.ecu.gui.androbd/log/AndrOBD.log.0.txt.lck
 E AndrOBD : 	at sun.nio.fs.UnixFileSystemProvider.newFileChannel(UnixFileSystemProvider.java:183)
 E AndrOBD : 	at java.nio.channels.FileChannel.open(FileChannel.java:292)
 E AndrOBD : 	at java.nio.channels.FileChannel.open(FileChannel.java:345)
 E AndrOBD : 	at java.util.logging.FileHandler.openFiles(FileHandler.java:460)
 E AndrOBD : 	at java.util.logging.FileHandler.<init>(FileHandler.java:403)
 E AndrOBD : 	at com.fr3ts0n.ecu.gui.androbd.MainActivity.setupLoggers(MainActivity.java:1787)
 E AndrOBD : 	at com.fr3ts0n.ecu.gui.androbd.MainActivity.onCreate(MainActivity.java:564)
 E AndrOBD : 	at android.app.Activity.performCreate(Activity.java:9013)
 E AndrOBD : 	at android.app.Activity.performCreate(Activity.java:8991)
 E AndrOBD : 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1531)
 E AndrOBD : 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3986)
 E AndrOBD : 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:4184)
 E AndrOBD : 	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:114)
 E AndrOBD : 	at android.app.servertransaction.TransactionExecutor.executeNonLifecycleItem(TransactionExecutor.java:231)
 E AndrOBD : 	at android.app.servertransaction.TransactionExecutor.executeTransactionItems(TransactionExecutor.java:152)
 E AndrOBD : 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:93)
 E AndrOBD : 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2602)
 E AndrOBD : 	at android.os.Handler.dispatchMessage(Handler.java:107)
 E AndrOBD : 	at android.os.Looper.loopOnce(Looper.java:232)
 E AndrOBD : 	at android.os.Looper.loop(Looper.java:317)
 E AndrOBD : 	at android.app.ActivityThread.main(ActivityThread.java:8623)
 E AndrOBD : 	at java.lang.reflect.Method.invoke(Native Method)
 E AndrOBD : 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
 E AndrOBD : 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:894)

## Expected behavior

Have any files created by AndrOBD

## AndrOBD Debug log files

Impossible because of the error

## Important details on your environment:

- Android OS version: Android 14
- AndrOBD version: 2.7.0 and 2.6.12
- OBD adapter type: not related
- Connected vehicle make, type, year, engine size: not related

fr3ts0n, 2024-07-18:
At initial startup AndrOBD does ask for R/W permissions. 
If you do NOT grant these permissions, then AndrOBD will NOT be able to write log files.
You should be able to grant file access at a later time in the "app info" dialog of Android

ieffuie4, 2024-07-18:
Of course i have granted these permissions like it asked for at first start. You can see the screenshot i have granted them in this bugreport: https://github.com/fr3ts0n/AndrOBD/issues/270
I was able to circumvent the issue by doing following:
Remove (uninstall) app.
Install app again but DO NOT START.
Go to android settings and search for the app. Then grant there BEFORE first start of the app all the rights the app want to have. Then start the first time the app. The app does not ask for any permissions and can write inside the whole android file system of the phone.

I think we can confirm that this is not some sort of a solution to the issue. The real solution would be not having to ask for this permissions like solved by other modern apps.