fr3ts0n/AndrOBD #272: Bug can not read any fault code

TVGSOFT, 2024-07-18:
# Describe the bug

I tried to use this app and read a fault code from my car, however I didn't receive any fault code from 3 modes: 03, 07, and 0A

But I used another app https://play.google.com/store/apps/details?id=com.ovz.carscanner&hl=en
And I can see a fault code. I tested on Mazda CX5 2020 as the same issue as well.
[image]

Is there any special method to read the fault codes?

## Important details on your environment:

- Android OS version: 12
- AndrOBD version: 2.6.12
- OBD adapter type: Bluetooth
- Connected vehicle make, type, year, engine size: KIA Sportage 2022 Gasoline 

## Additional context

Add any other context about the problem here.

fr3ts0n, 2024-07-18:
Please could you attach a log file?
See: https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues#debug-log-files

Thx

TVGSOFT, 2024-07-18:
[AndrOBD.log.0.txt](https://github.com/user-attachments/files/16283690/AndrOBD.log.0.txt)
@fr3ts0n This is the log.

ieffuie4, 2024-07-18:
@TVGSOFT 
You missed to configure the debug level.
Quote from the link that describe how to access the logfiles: "Preferred detail level for issue analysis is FINE."
But you have kept the default log level instead of setting it to FINE.

TVGSOFT, 2024-07-19:
[AndrOBD.log.0.txt](https://github.com/user-attachments/files/16300390/AndrOBD.log.0.txt)
@fr3ts0n Attached the log.

fr3ts0n, 2024-07-20:
None of the logged vehicle responses does indicate any DFC set.
All 3 OBD services of DFC reading report xx00, which means NO fault set.
Also the OBD data service reports 0 faults set.

Does the other app read additional NON-OBD services to read failure codes?

TVGSOFT, 2024-07-22:
Hi @fr3ts0n 

I understand that they test the hidden DTC base on a mechanism.
It's not part of OBD II mode.
Do you have any experience on this?

fr3ts0n, 2024-07-22:
No, not really. 
If it's not detected / reported in OBD services there is not much chance to add this to AndrOBD.

fr3ts0n, 2024-07-22:
Sorry, I forgot to attach the picture of the s/n tag.
This was also attached to the service case.
-- 
Best Regards
Erwin Scheuch-Heilig