fr3ts0n/AndrOBD #273: App is writing into the MAIN AndroidOS directory instead of /Android/data/com.fr3ts0n.ecu.gui.androbd/

ieffuie4, 2024-07-18:
# Describe the bug

Now i understand why AndrOBD ask for such a deep file access permissions and cant work without it. Its writing in the MAIN Android directory. This is really nothing a app should do. This should really be fixed. Doing so is for sure just wrong.
After i did this https://github.com/fr3ts0n/AndrOBD/issues/271#issuecomment-2236829106 i gave the AndrOBD really deep file access permissions. Now AndrOBD could in theory delete my whole phone, read all my pictures, access the files of every other app, and so on.
AndrOBD wants for no reason to create a directory here:
/com.fr3ts0n.ecu.gui.androbd
Not where it should be: /Android/data/com.fr3ts0n.ecu.gui.androbd/

You do not have to ask the user/AndroidOS for granting additional file access permissions when you want to write into the app-own directory: /Android/data/com.fr3ts0n.ecu.gui.androbd/
Please take a look into this Android 5.0+ app sourcecode that do NOT ask the user for any additional file access permissions: https://github.com/libre-tube/LibreTube

## To Reproduce

https://github.com/fr3ts0n/AndrOBD/issues/271#issuecomment-2236829106

## Expected behavior

Do not ask for any file access permissions and just use the default directory /Android/data/

## Important details on your environment:

- Android OS version: 14
- AndrOBD version: 2.7.0
- OBD adapter type: not related
- Connected vehicle make, type, year, engine size: not related

serppower, 2024-10-15:
Is this fixed?

fr3ts0n, 2024-10-16:
No, issue is still open