fr3ts0n/AndrOBD #277: Flashlight button inside the app

aha999, 2024-08-30:
**Is your feature request related to a problem? Please describe.**
When doing car diagnostics, users in my opinion tend to use their phone's flashlight a lot, so currently the users have to exit the app and launch the flashlight outside the app.

**Describe the solution you'd like**
It would be really useful if users of AndrOBD had a flashlight toggle button inside the app, that would activate the phone's flashlight, therefore aid a lot in finding the OBD port in the dark spot, and whereever the flashlight comes in handy when doing car diagnostics.

**Additional context**
This would be helpful cool feature that users would probably appreciate, myself included.

fr3ts0n, 2024-09-11:
Sorry, but I don't think this is a big improvement. 
At least on all the phones I use, I can set a hardware key/button to toggle the flash light. This is way more flexible and valid for any app running ...

ieffuie4, 2024-09-19:
I also vote against this. On modern phones you just put the finger on top of the screen while running AndrOBD, pull down the buttons and press flashlight toggle. This works on nearly all android apps that do not fully force fully screen mode and even there it often works. It would be full waste of dev-time to add such a thing.

closed 2026-02-27 by fr3ts0n