fr3ts0n/AndrOBD #278: How to connect this OBD2 device without bluetooth?

VasilKotsov, 2024-09-04:
Hello guys I have this device is that missing bluetooth.
I want to connect it to this app instead of official app, because the official app requires phone number.

[image]
[image]
[image]
[image]

fr3ts0n, 2024-09-11:
You can give it a try connecting via USB. But if it doesn't support ELM protocol you will not be successful.
As described the APP is designed to talk to ELM adapters.

VasilKotsov, 2024-09-11:
I try to connect via USB and the app is showing the device, but when I try to connect the App crash.
[image]

VasilKotsov, 2024-09-12:
[AndrOBD.log.0.txt](https://github.com/user-attachments/files/16984071/AndrOBD.log.0.txt)
Also there is a another file but the type is TXT.lck
The device is connected only with my phone over usb but is not plugged on the car.

VasilKotsov, 2024-09-12:
[AndrOBD.log.1.txt](https://github.com/user-attachments/files/16984126/AndrOBD.log.1.txt)