fr3ts0n/AndrOBD #279: How do I know when the drive cycle is complete?

Geremia, 2024-09-11:
After resetting fault codes, how do I know when the drive cycle is complete?

fr3ts0n, 2024-09-11:
You don't. Generally a drive cycle goes from turning the ignition on to turn the ignition off after a ride. 
However, in context of clear codes the drive cycle to re-learn values does typically take several minutes. You may have to drive some meters to have some load condition on the motor and to get it to regular operating temperatures.

Geremia, 2024-09-12:
@fr3ts0n  The reason I ask is that I have an error code (P0456) with a ⚠️ symbol by it, and below it is a P0000 error code that says no error codes set; however, the engine check light is off. Will the error code with the ⚠️ go away eventually? (Also, it's not in the pending section.)