fr3ts0n/AndrOBD #282: Android 14 update.

MrMayonnaisez, 2024-10-29:
Hello, I would like to use this software due to its open source nature however I'm limited by the fact it is not made for Android 14 which my device uses. Is there an eta on any update or any plans to update the app?

Currently I'm using OBD AutoDoctor and I don't like it because it hides information from sensors behind a paywall.

fr3ts0n, 2024-11-02:
Since I currently do not own a Android 14 phone, I can't test it on the real device. 
However, it does run without a problem on the Android 14 emulation.

Please could you describe the problems you have with it on Android 14?
Please also could you add log files where helpful.
Thx

r100gs, 2024-11-20:
Hello @fr3ts0n, 

I tested it today on android 14 and the app loads but does not display something on screen.

Best regards,
Stefan

fr3ts0n, 2024-11-24:
Do you have a logfile / android logcat output of this behaviour?
Thx

VoronManiac, 2024-12-15:
Same.  Trying to install from Android Play Store and getting message this app isn't available on your device because it was made for an older version of Android (current is version 15).  It would be a shame to let all the hard work that went into this fall by the wayside...Is a rebuild possible or are there fundamental design issues?

fr3ts0n, 2024-12-20:
The version on google play store is rather old (V2.03) and NOT maintained by it's owner.
I have already asked cppox (the user who owns the playstore release) to remove it, since it also includes ads, which I do NOT want at all. -> No reaction. 
Since Cppox doesn't provide (changed) source code, I filed a license infringement to Google, but this also was ignored.
So much to the organizational fun of an OSS developer. 😢 

However on the technical side there are several design issues which keep me from updating the target versions of the project:
- permissions handling changes drastically between SDKs
  - Each SDK version update would have created additional (rather ugly) workarounds for changed handling of permissions on
    - Bluetooth
    - USB
    - Files (Measurements, CSV exports, logs)
- plugin communication does break on target sdk set >25
  - IPC communication handling changed

The current version (V2.6.13) is running fine on an Android 14 emulator target (with current SDK settings), but:
- f-droid update is blocked because of app's SDK settings
  - As a workaround it can be downloaded from f-droid web and installed manually
- APP shows a warning message at initial startup
  - needs to be confirmed
- Permissions have to be accepted "the old way" at initial startup
  - needs to be confirmed by dialog

I currently struggle to find enough motivation for these adaptions to happen 😩