fr3ts0n/AndrOBD #283: No data/connection loop

404invalid-user, 2024-11-20:
# Describe the bug

A clear and concise description of what the bug is.

## To Reproduce

Steps to reproduce the behavior:

1 open app 
2 turn on car 
3 plugin ELM
2.5 also tried steps 3 then 2
4 click OBD data 

## Expected behavior

A clear and concise description of what you expected to happen.

see my cars obd2 data 

## AndrOBD Debug log files

Please attach AndrOBD debug log files.

AndrOBD stores log files in a subdirectory on your primary storage device. (com.fr3ts0n.ecu.gui.androbd/log/AndrOBD.log.*)

[AndrOBD.log.0.txt](https://github.com/user-attachments/files/17834694/AndrOBD.log.0.txt)

### Log detail level

The detail level of the logged debug information can be adjusted in the Settings > Development options > Logging level.

By default the detail level is set to INFO.
Preferred detail level for issue analysis is DEBUG. (Since V1.5.0 this level is called FINE)

ALL

## Important details on your environment:

- Android OS version: 14
- AndrOBD version: latest V2.6.12
- OBD adapter type: ELM327 Bluetooth 
- Connected vehicle make, type, year, engine size: Renault Clio v 2023 1L

## Additional context

Add any other context about the problem here.

404invalid-user, 2024-11-20:
Using another app to manually set the preferred OBD protocol I have fixed the connecting disconnected loop but now selecting items it says no data 
[image]

fr3ts0n, 2024-11-24:
Thank you for your report,
It looks like your vehicle does have a "interesting" implementation of the OBD protocol.
It does respond a additional byte 0xAA on the request of available data items for OBD data (0100) and vehicle information (0900).
This definitely violates the OBD spec. 
However I'll try to find a fix for that, since improves the robustness of the app. 👍

404invalid-user, 2024-11-24:
typical french haha. thank you for taking a look at it.

fr3ts0n, 2024-11-24:
Beta version is available for testing here:
https://github.com/fr3ts0n/AndrOBD/actions/runs/11998890153/artifacts/2230122269

404invalid-user, 2024-11-24:
did a quick test and it looks to be working perfectly thank you 
[image]
[image]

commit 7bf7133f (2024-11-24): Fix #283 - Handle additional padding data
closed 2025-01-06 by fr3ts0n