fr3ts0n/AndrOBD #285: AndrOBD crashes when establishing USB connection to ELM327 adapter

20210213, 2024-12-14:
# Describe the bug

When establishing the USB connection in the usual way, recent versions of AndrOBD crash as follows:

Steps to reproduce the behavior:
1. plug USB cable from the adapter into the cellphone
2. confirm to open AndrOBD
3. The proposed USB connection is displayed. Click on it to confirm.
4. Android displays that AndrOBD has crashed.

## Expected behavior

Instead of crashing according to item 4, AndrOBD should open and communicate with the ELM327 adapter.

## AndrOBD Debug log files

Please attach AndrOBD debug log files.
[AndrOBD.log.1.txt](https://github.com/user-attachments/files/18138061/AndrOBD.log.1.txt)

## Important details on your environment:

- Android OS version: 9
- AndrOBD version: 2.6.13
- OBD adapter type: noname ELM327 USB adapter

## Additional context
Note: When no USB connection is established, e. g. with no cable connected to the cell phone, AndrOBD 2.6.13 will start. 

The last version which worked OK with the same adapter was  2.6.7.

We encountered an identically looking issue already in February 2021; it was fixed after a bug report.

fr3ts0n, 2024-12-15:
Thank you for your report,
I can reproduce the problem on connecting to my USB adapter. 
(Pls remind me to include this as a smoke test for releases)

I can see following error message in the android logcat output when the USB device class is instantiated:
``` java.lang.NullPointerException: Attempt to invoke virtual method 'int android.hardware.usb.UsbEndpoint.getMaxPacketSize()' on a null object reference```  

I'll contact @mik3y who is the developer of the USB device library used by AndrOBD. 
Reverting the library to the version 3.3, which was used with V2.6.7 (your reported last known good version) would fix the problem.
However, there are several extensions and fixes listed in the changelog of the library.

20241229-hub, 2024-12-29:
Unfortunatly V2.6.7 is not available any more on F-Droid until last release on 09.12.2024. Only last 3 versions are provided. I need this V2.6.7 apk to run my USB-ELM327 to get a car problem identified. It worked well 8 month ago. Can you please provide the executable (apk) on F-Droid, github or by mail.

fr3ts0n, 2025-01-01:
In fdroid APP you can enable the archive in repo settings. This allows you to install historical versions.

20241229-hub, 2025-01-01:
Many thanks for your hint. I did find the little bit hidden function in f-droid. It worked well, the car ist checked.--Gesendet mit der WEB.DE Mail App

TheBagarius, 2025-01-05:
I'm having the same error on my android head unit

# Important details on your environment:

    Android OS version: 10
    AndrOBD version: 2.6.13
    OBD adapter type: chinesium ELM327 USB adapter

never attempted to use 2.6.7 so cannot say if that will work.
I will dump the debug files from the car tomorrow and share tomorrow / look into where the issue might be.

looking forward to working with everyone to get this fixed (and more in the future)

20210213, 2025-01-06:
This looks good:
1. plug USB cable from the adapter into the cellphone
2. confirm to open AndrOBD
3. The proposed USB connection is displayed. Click on it to confirm.
4. AndrOBD displays that the USB connection has been established (and the adapter starts to flash).
This was a fast, however preliminary test because the ELM327 adapter wasn't connected to a car; should be properly confirmed.
Nevertheless - thank you!

20210213, 2025-01-09:
> This looks good:
>
>     1. plug USB cable from the adapter into the cellphone
>
>     2. confirm to open AndrOBD
>
>     3. The proposed USB connection is displayed. Click on it to confirm.
>
>     4. AndrOBD [...]

This is to confirm that the test version https://github.com/fr3ts0n/AndrOBD/issues/285#issuecomment-2573272232 works properly. The cell phone connects via usb to the ELM327 adapter which transfers the readout data from the car.
Thank you for fixing this issue.

closed 2025-01-10 by fr3ts0n