fr3ts0n/AndrOBD #288: AndrOBD asks for ECU address

PinkFreud, 2025-05-18:
Using a 'KMOON' branded ELM327 bluetooth adapter to connect to a 2023 Hyundai Ioniq 5, AndrOBD asks me to select from one of several addresses:

[image]

I'm not certain which address to use here, nor am I certain if using the wrong one can cause issues.

AN0MSU, 2025-05-18:
Hi,

Are you located in Adelaide South Australia?
If so I can take a look at the problem for you at my workshop. Let me know.

Ian

fr3ts0n, 2025-05-18:
Looks like there are multiple ECU's answering the OBD requests. Worst case you'll have to select one by one and check OBD info for each to talk to the right one ...

PinkFreud, 2025-05-19:
I wasn't sure if that was safe to do, but I'll give it a try.

Thanks!

closed 2025-07-23 by fr3ts0n