fr3ts0n/AndrOBD #290: Install failed on Android 15

anno0, 2025-07-24:
# Describe the bug
Unable to install on Anroid 15

## To Reproduce

Steps to reproduce the behavior:
1. Install F Droid
2. Select AndrOBD to install

Installation process failes with "unkown error" 

## Expected behavior

It should be installed.

## Important details on your environment:

- Android OS version: 15 AP3A.200905.015.A2
- AndrOBD version: Latest on FDroid
- OBD adapter type:
- Connected vehicle make, type, year, engine size:

## Additional context

Add any other context about the problem here.

fr3ts0n, 2025-08-08:
Do you have any additional information on this problem?

Could it be worth a try to download apk from fdroid and install via adb?
At least this should give more information...

Thx

anno0, 2025-08-22:
I'm sorry. 
I updated FDroid last day. Now I get a message, that your AndrOBD version is too old for Android 15.

I also tried to install your beta apk file. Here I get the following message. It can' be installed.

[image]

fr3ts0n, 2025-09-13:
New attempt:
- added declaration of permission
  - even if only used/required implicitly ba SDK

Fix is contained in new beta V2.6.16:
https://github.com/fr3ts0n/AndrOBD/actions/runs/17693525334

Pls could you test this again, since my virtual test devices even accepted the install before the change ...
Thx

CoolCrescent186, 2025-10-28:
I'm on LineageOS 22.2 (Moto G100), which is Android 15, and AndrOBD installs and runs without issue.

jfu334, 2026-01-03:
I ran into the same problem on a Fairphone 6 with Android 15. Apparently this was caused by google play protect, after disabling it the installation worked.